gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #1: Guide (semi noob but not entirely)

Frikkie50, 2023-03-02:
Loving the fact that i can pull info no.

Do you have a guide on how to use the loadshedding / sunsynk timer changes? 

Love the work you have done.

gdwaterworth, 2023-03-03:
Still a very new project and the automation atm is undergoing alot of change. 
There are a few dependencies on other addon/integrations.
Once things are a bit more stable and it is a bit more cleaned up I will document it

Frikkie50, 2023-03-06:
Thanks man, loving what you have done. Lately the sunsynk site doesnt work for while

gdwaterworth, 2023-03-07:
The e-linter site is down now and again for a few hours at a time. Trying to build in more error checking for it, but it is hard :/ 
Yesterday the "settings" were not being updated for about 3 hours.

Frikkie50, 2023-03-29:
Perhaps one day they will open up the dongle to allow us to grab information from there

CraigBoswell, 2023-03-29:
> Perhaps one day they will open up the dongle to allow us to grab information from there

So agree with you. I think I read somewhere it is open to Phillip's Hue, but I don't have one of those. Maybe if enough users pester Sunsynk they'll open it.

gdwaterworth, 2023-03-29:
Pfft not holding my breath.

Frikkie50, 2024-01-27:
any guide on doing v2.3?

Frikkie50, 2024-01-29:
> any guide on doing v2.3?

Just ignore my comment got it installed. Clearly I missed the short guide.

Frikkie50, 2024-03-06:
Thanks for all the hard work on this.

closed 2024-03-06 by Frikkie50