gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #11: Basic write to Sunsynk for Timer Mode On/Off

CraigBoswell, 2023-04-03:
Hi there, I am ultimately wanting to do more sophisticated writes to the Sunsynk schedule, but initially keen to just change System Mode from use Timer to Manual (full batt) in the case of pending load-shedding and low battery.
Is it possible with your current integration to do that, or is it part of the dev you are working on?
If that component could work it would be a great step forward in the load-shedding vs battery SOC management.
Much appreciate your work, the data is flowing very reliably.

gdwaterworth, 2023-04-03:
There is a portion under the automation flows that effectively deals with that. 
BUT I NEED TO REITTERATE THAT THESE ARE NOT NEW GUY FRIENDLY at the moment. 
The problem is that there is very little information ( As in Zero ) from api documention I have. 

Basically I have started documenting in this wiki page all the variables that are available. There are more , just have not had time to document yet:

https://github.com/gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk/wiki/E-Linter-Information#variables-available-for-update

Basically to set a settings you need to do a http(s) web with a crafted json payload 

{
sn: xxxxx,
parameter1:data1,
parameter2:data2
}

The corrected headers ( with bearer code ) and url needed are configured allready in global variables

```
msg.headers = global.get("SS.Header.Bearer");
msg.url = global.get("SS.url.i1.set");

```

Heindelange, 2023-07-18:
Thank you.  This helped me schedule my work mode.

I wanted to set my work mode to to export to home (incl non essentials) when there is sun, but didn't want my geyser to drain my battery every evening, so changed it back to Limit to load in the afternoon.

I used a time picker card which seems to work nicely so I don't need to hard-code my time in node-red.

These were the parameters required:
sysWorkMode: 1 - Zero Export - Limit to load
sysWorkMode: 2 - Limit to Home (incl non-essentials)

Kabz777, 2023-08-31:
> 

Hi, how did you go about doing this? I can't seem to read/write the Use Timer setting or the System Mode settings.
Thanks

gdwaterworth, 2023-12-12:
Some basic settings update included in v2

Kabz777, 2023-12-12:
> Some basic settings update included in v2

Noob here, is there seamless way to change from. V1 to V2 that will keep the sensors, history, etc.?

gdwaterworth, 2023-12-13:
There are only a few entities that have statistics enabled
They reference another entity and because we have not changed those stats sensors they should retain their data.
Make a backup of your system before you do the changes.

The rest are only 7 days (?) depending on settings so I don't factor those into my thinking.

Just check you have no others.

##### Misc Sensors for Energy
  - sensor:
      - name: "Sunsynk Battery etotalchg"
        state_class: total_increasing
        unique_id: "sensor.sunsynk_battery_etotalchg"
        device_class: energy
        unit_of_measurement: kWh
        state: >
            {{state_attr('sensor.sunsynk_usable_battery', 'etotalchg')|float}}
  - sensor:
      - name: "Sunsynk Battery etotaldischg"
        state_class: total_increasing
        unique_id: "sensor.sunsynk_battery_etotaldischg"
        device_class: energy
        unit_of_measurement: kWh
        state: >
            {{state_attr('sensor.sunsynk_usable_battery', 'etotaldischg')|float}}
  - sensor:
      - name: "Sunsynk PV Total"
        state_class: total_increasing
        unique_id: "sensor.sunsynk_pv_total"
        device_class: energy
        unit_of_measurement: kWh
        state: >
            {{state_attr('sensor.sunsynk_usable_pv', 'etotal')|float}}
  - sensor:
      - name: "Sunsynk Grid Total"
        state_class: total_increasing
        unique_id: "sensor.sunsynk_grid_total"
        device_class: energy
        unit_of_measurement: kWh
        state: >
            {{state_attr('sensor.sunsynk_usable_grid', 'totalusedin')|float}}
  - sensor:
      - name: "Sunsynk Load Total"
        state_class: total_increasing
        unique_id: "sensor.sunsynk_load_total"
        device_class: energy
        unit_of_measurement: kWh
        state: >
            {{state_attr('sensor.sunsynk_usable_load', 'totalused')|float}}

Kabz777, 2023-12-13:
Ah makes sense, thanks! And thanks for all the efforts!