gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #12: Request for Data // Some information

dj-louw, 2023-04-11:
Hi, 

On your wiki you have requested some data from users. I can provide you with some of this.

I have a geyser/hot water cylinder connected to my Aux port of the inverter, which is set to come on only if the battery is fully charged and if there is sufficient solar to do so.

At this time, I do not have anything connected as non-essential loads on the grid side of the inverter.

As an aside, I don't think the inverter can actually report on the Aux port load. I am not seeing that value on the app anywhere, and neither am I seeing it in any of the below datasets. On the inverter screen itself, it reports Aux separately, but then it _includes_ the aux load when showing the UPS load as well. My geyser is 2kW, the home draws 500W. So Aux port reads 2kW, UPS port reads 2.5kW.

You have requested info about the following. See my provided data below.

## `sensor.sunsynk_load`

```
totalUsed: 73.9
dailyUsed: 5.2
vip_0_volt: '243.3'
vip_0_current: '0.0'
vip_0_power: 4833
totalPower: 4833
smartLoadStatus: 1
loadFac: 49.96
upsPowerL1: 0
upsPowerL2: 0
upsPowerL3: 0
upsPowerTotal: 0
total: 73.9
volt: '243.3'
power: 4833
totalpower: 4833
calc_current: '19.86'
```

## `sensor.sunsynk_input`

```
pac: 4703
pvIV_0_id: Missing
pvIV_0_pvNo: 1
pvIV_0_vpv: '388.7'
pvIV_0_ipv: '12.3'
pvIV_0_ppv: '4703.0'
pvIV_0_todayPv: '0.0'
pvIV_0_sn: '2209244812'
pvIV_0_time: '2023-04-11 10:24:56'
pvIV_1_id: Missing
pvIV_1_pvNo: 2
pvIV_1_vpv: '2.5'
pvIV_1_ipv: '0.0'
pvIV_1_ppv: '0.0'
pvIV_1_todayPv: '0.0'
pvIV_1_sn: '2209244812'
pvIV_1_time: '2023-04-11 10:24:56'
mpptIV: Missing
etoday: 9.3
etotal: 84.8
pv: 4703
pv1: 4703
pv1_volt: 388.7
pv1_current: 12.3
pv2: 0
pv2_volt: 2.5
pv2_current: 0
pvtotal: 84.8
```

## `sensor.sunsynk_grid`

```
vip_0_volt: '243.1'
vip_0_current: '2.6'
vip_0_power: 0
pac: 0
qac: 0
fac: 50.06
pf: 1
status: 0
etodayFrom: '2.7'
etodayTo: '0.0'
etotalFrom: '10.5'
etotalTo: '0.0'
limiterTotalPower: -40
total: 10.5
volt: '243.1'
current: '2.6'
gridonline: true
power: 0
```

## `sensor.sunsynk_output`

```
vip_0_volt: '241.3'
vip_0_current: '19.4'
vip_0_power: 4833
pInv: 4703
pac: 4833
fac: 50
```

## `sensor.sunsynk_flow`

```
custCode: 29
meterCode: 0
pvPower: 4703
battPower: 408
gridOrMeterPower: 0
loadOrEpsPower: 4833
genPower: 0
minPower: 0
soc: 99
pvTo: true
toLoad: true
toGrid: false
toBat: false
batTo: true
gridTo: false
genTo: false
minTo: false
existsGen: false
existsMin: false
genOn: false
microOn: false
existsMeter: false
bmsCommFaultFlag: false
pv: Missing
existThinkPower: false
```

dj-louw, 2023-04-11:
Additionally, I am registered on Region 2. So let me know if I can help with testing or information gathering.

gdwaterworth, 2023-04-12:
> Additionally, I am registered on Region 2. So let me know if I can help with testing or information gathering.

I have added a version for Region 2 That should work

closed 2023-12-12 by gdwaterworth