gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #22: Multi-inverter config?

antiphotom, 2023-08-02:
Hi, love this as it's the first way I've been able to get data from my Sunsynk system into HA.  Thank you so much!

Have you looked into a configuration for multi-inverter systems?  I have two 8kW Ecco units running in parallel and my initial testing suggests it's only pulling data from one inverter.  I could just double all the values but it would be neater/more useful to be able to see the data from both inverters independently.

Happy to collaborate so that you can see my data if that helps.

Cheers from the UK!

uwan-cm, 2023-11-21:
That would be great!  I also have two inverters in parallel.

gdwaterworth, 2023-12-12:
Basis Functionality included in V2

closed 2023-12-12 by gdwaterworth