gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #24: Change from Region 1 to Region 2

dharmeshbhana, 2023-12-04:
Hi

I have the automation installed for region 1. Sunsynk have been transferring everyone to region 2.

1) Does this integration work for the New Sunsynk migration to region 2 for everyone?
2) if so, is there a quick fix for region 2 or must the a new Node Red flow be installed for region 2.

Thanks

wacalitz, 2023-12-04:
I only use the gather flow section and just had to change the following lines (87 & 88) in the Setup Global Vars

Just swap to https://api.sunsynk.net instead of https://pv.inteless.com (see below)

`    // Sunsynk V2 extra info 
    // global.get("SS.url.hostname")
    global.set("SS.url.hostname","https://api.sunsynk.net") ;
    // global.set("SS.url.hostname","https://pv.inteless.com");
`

Hope this helps

dharmeshbhana, 2023-12-04:
Great, thanks for your help, Appreciated. Just to confirm (for us not so technical). Below extract from my code. Just to change the  pv.inteless.com to https://api.sunsynk.com, everything else remains the same

    var headers_with_bearer = {
        Authorization: "Bearer " + global.get("SS.Bearer"),
        "Content-Type": "application/json",
        "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36",
        "Connection": "keep-alive",
        "Accept": "*/*",
        "Cache-Control": "no-cache",
        "Origin": "https://sunsynk.net",
        "Referrer": "https://sunsynk.net"
    }
    global.set("SS.Header.Bearer", headers_with_bearer);
    global.set("SS.url.hostname", "**https://pv.inteless.com**");
    // Sunsynk V2 extra info 
    // global.get("SS.url.hostname")
    // global.set("SS.url.hostname","https://api.sunsynk.net") ;
    node.status(global.get("SS.url.hostname"));

wacalitz, 2023-12-04:
I just removed the // in front of the line with https://api.sunsynk.net and add // before the line with https://pv.inteless.com

Below is an extended extract of mine, just remember to hit Done and Deploy at the top as well to save and activate the changes.

>   
    var headers_with_bearer = {
        Authorization: "Bearer " + global.get("SS.Bearer"),
        "Content-Type": "application/json",
        "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/103.0.0.0 Safari/537.36",
        "Connection": "keep-alive",
        "Accept": "*/*",
        "Cache-Control": "no-cache",
        "Origin": "https://sunsynk.net",
        "Referrer": "https://sunsynk.net"
    }
    global.set("SS.Header.Bearer", headers_with_bearer);

    // Sunsynk V2 extra info 
    // global.get("SS.url.hostname")
    global.set("SS.url.hostname","https://api.sunsynk.net") ;
    // global.set("SS.url.hostname","https://pv.inteless.com");

dharmeshbhana, 2023-12-04:
You're such a star. Thank you. Works perfectly. :-)

gdwaterworth, 2023-12-12:
It is now a configuration item in V2

closed 2023-12-12 by gdwaterworth