gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #44: Spelling for settings values

Gr3mlin13, 2024-01-01:
Hi,

Thanks for the great work so far, got me going on setting up my inverter with Home Assistant. I noticed a bug that would make the setting of charging times on and off very inconsistent. The spelling when setting the values matters, in Postman I teste that "time1On" would work, but "time1on" would not for example. I have it consistently setting zones to charge from grid when loadshedding is two hours away, and just using solar when no loadshedding is expected after changing the spelling. 

I am happy to share the changes I made back if that would help you. It is working fine for now, but will be dissecting it a bit further once I have the time. happy to conbtribute the learnings back if you need, otherwise I will just fork and make my changes there.

Thanks,

gdwaterworth, 2024-01-01:
Need to understand what you are changing. Yes case matters when sending the update to sunsynk. 
Are you using the input boolean fields that were created for end user use ?

Gr3mlin13, 2024-01-02:
Okay, to be more specific. In your NodeRed flows you use "timeXon", with a lowercase. This is consistent with the payload recieved from them when calling settings read.

However, this was not giving me consistent results, so I tested the following payloads to narrow down the issue, with my results:

- Payload1 Lowercase O : Result no changes to setting
{
    "time1on": "1"
}
- Payload2 Uppercase O : Results, changes the setting
{
    "time1On": "1"
}

From my testing it looks like the API is not consistent in what it returns and what it expects.

Also note that I did not include genTimeXon in any payloads, or anything else. Those are my full payloads in Postman. It may be that either the combination or the other spelling will get the job done.

My changes were in "Setup Global Vars" and "Export Data Export To Entity" to change from timeXon to timeXOn, and is working consistenly now with the rest of your code.

gdwaterworth, 2024-01-02:
Hi
Sorry 1st , What is postman ?

You cannot just change Global Vars and Export Data as those setups are used in a variety of places
It is also used in the mapping and pairing tables to determine when fields have changed and what needs to be set in the inverter.

My testing so far has been consistent and you need to take into account the requirements when setting that variable and updating the inverter.

Also you need to take into account that there is a deliberate time delay inserted to allow time for the inverter to actually get the change and implement it. It is not realtime and can take up to a minute ( or longer ).
If you do changes before that timer runs out then you may see inconsistent results. That timer is there for a reason

timeXon has always been lowercase coming from the api and submitting to the api.
There are others where there is a capital in the name, but the timeXon is not one of them

Regards

gdwaterworth, 2024-01-02:
In addition what region are you on ?

gdwaterworth, 2024-01-02:
Also note that I did not include genTimeXon in any payloads, or anything else.

This is why your testing is inconsistent
That is a requirement from my  testing

Did you also exclude the sn: from your payload ?

Gr3mlin13, 2024-01-02:
Postman is a tool to build and test APIs. I use it to query call the API directly, removing the variable of NodeRed to test just one thing at a time. I find it much quicker to work in there and figure out the API, as I am not yet that proficient in NodeRed yet.

I was simplifying, but you use the variables set in global to build your payload. When I changed those after my testing, I noticed the settings from the inverter were not updating correcly anymore, so I updated the Export Data entity to match the changes in Global Vars.

I am aware of the timer, I was waiting for confirmation that the post has happened anbd the reply recieved before checkning on the inverter that the settings have changed. Same process as when testing with the API directly.

On region 2 as everyone else after the migration. 

As I said above, I get consistent results with the entire payload being just the one option, no genXon, no sn:
{
    "time1On": "1"
}

Gr3mlin13, 2024-01-02:
This is howe I tested and figured it out. I started with sending back the variables recieved the api call GET : https://api.sunsynk.net/api/v1/common/setting/XXXXXXXXXX/read:
`
{
	"code": 0,
	"msg": "Success",
	"data": {
		"sellTime1": "02:00",
		"genTime2on": false,
		"beep": 0,
		"sellTime2": "06:00",
		"wattOverExitFreqStopDelay": 0,
		"sellTime5": "18:00",
		"sellTime6": "22:00",
		"sellTime3": "10:00",
		"sellTime4": "14:00",
		"exMeterCtSwitch": 0,
		"sdChargeOn": 1,
		"lockInVoltVar": "0.0",
		"time2on": false,
		"batWarn": 0,
		"wattVarEnable": 0,
		"reconnMinVolt": 0,
		"caFStart": 0,
		"pvMaxLimit": 3000,
		"sensorsCheck": -1,
		"underFreq2Delay": 0,
		"varQac2": "0",
		"varQac3": "0",
		"varQac1": "0",
		"wattUnderExitFreq": 0,
		"overVolt1": 0,
		"overVolt2": 0,
		"varQac4": "0",
		"genPeakPower": 8000,
		"meterB": 0,
		"eeprom": 0,
		"meterA": 0,
		"comSet": 3819,
		"caVoltPressureEnable": 0,
		"meterC": 0,
		"wattUnderFreq1": 0,
		"solarMaxSellPower": 6500,
		"acCoupleOnGridSideEnable": 0,
		"thursdayOn": true,
		"time3On": 0,
		"batteryRestartCap": 35,
		"overFreq1Delay": 0,
		"bmsErrStop": 0,
		"checkTime": 10,
		"atsSwitch": 0,
		"acCurrentUp": 0,
		"sacPowerControl": 0,
		"rsd": 0,
		"batteryOn": 1,
		"genTime1on": false,
		"volt12": 490,
		"volt10": 550,
		"volt11": 490,
		"time1on": false,
		"wattUnderFreq1StartDelay": 0,
		"rcd": 0,
		"chargeVolt": 56,
		"wednesdayOn": true,
		"floatVolt": 55.2,
		"workState": 0,
		"loadMode": 0,
		"sysWorkMode": 2,
		"sn": "XXXXXXXXXX",
		"genCoolingTime": 0,
		"genPeakShaving": 0,
[... 264 lines cut ...]
		"sellTime2Volt": 49,
		"arcFactF": 884630023,
		"arcFactI": -383828332,
		"acVoltLow": 0,
		"genSignal": 0
	},
	"success": true
}
`

Checking this now, it seems like both are available in the response. It might be that timeXOn works on its own, while timeXon needs the other variables to be consistent. I just used the uppercase one as 1 and 0 are shorter to type than true and false.
`
"time2on": false,
"time3On": 0,
"time1on": false,
"time2On": 0,
"time4on": false,
"time5On": 0,
"time3on": false,
"time4On": 0,
"time6on": false,
"time5on": false,
"time6On": 0,
"time1On": 0,
`

gdwaterworth, 2024-01-02:
Hi

That is not the way that the sunsynk/pv.inteless.com website handle the api update
I try emulate them as close as possible hence why I do not use the O version
Also excluding the sn is dangerous in multi inverter/plant setups.

I am happy with the way the updates happen when I pair the gen and normal time, so will not be changing that

Regards

Gr3mlin13, 2024-01-02:
Okay quick test, it looks like the pairing works as well, but have to use true and false instead of the numbers.
{
    "genTime1on": true,
    "time1on": true
}

So can use the pairing with true or false, or just uppercase with 1 or 0. Will go and check why your default did not work for me.

> That is not the way that the sunsynk/pv.inteless.com website handle the api update
Do you have a reference for how the sunsynk website hadles it, would be much easier than the manual testing I have been doing so far?

gdwaterworth, 2024-01-02:
Yes , It has always been that way

I just checked the website submission again :
They only use :
time1on
   true
time2on
   true
time3on
   true
time4on
   true
time5on
   true
time6on
   true

The capital O being returned in the settings get call is something new
There was a problem a while back where the timeslot times were “broken” and is probably due to that
I wouldn’t use them

battMode
"-1"
cap1
"50"
cap2
"30"
cap3
"25"
cap4
"85"
cap5
"80"
cap6
"70"
energyMode
"1"
fridayOn
false
genTime1on
true
genTime2on
true
genTime3on
true
genTime4on
true
genTime5on
true
genTime6on
true
mondayOn
false
peakAndVallery
"1"
pvMaxLimit
"8000"
safetyType
"2"
saturdayOn
false
sellTime1
"00:00"
sellTime1Pac
"8000"
sellTime1Volt
"49"
sellTime2
"04:30"
sellTime2Pac
"8000"
sellTime2Volt
"49"
sellTime3
"05:00"
sellTime3Pac
"8000"
sellTime3Volt
"49"
sellTime4
"15:30"
sellTime4Pac
"8000"
sellTime4Volt
"49"
sellTime5
"22:00"
sellTime5Pac
"8000"
sellTime5Volt
"49"
sellTime6
"23:00"
sellTime6Pac
"8000"
sellTime6Volt
"49"
sn
XXXXXX
solarMaxSellPower
"9000"
solarSell
"0"
sundayOn
false
sysWorkMode
"1"
thursdayOn
false
time1on
true
time2on
true
time3on
true
time4on
true
time5on
true
time6on
true
tuesdayOn
false
wednesdayOn
false
zeroExportPower
"20"

Gr3mlin13, 2024-01-02:
Got it, would be helpfull if there was easily accesible API docs available. For now will keep using as is.

Thanks,

closed 2024-01-02 by Gr3mlin13

gdwaterworth, 2024-01-02:
You are wishing. What is being done here is not condoned by Sunsynk and done at your own risk

gdwaterworth, 2024-01-02:
Just to phrase it nicely

There are some “limited” api docs available which are available in the git repo
There is absolutely nothing on setting settings

Everything that has been done in this work is all reverse engineered from looking at the application traffic

Gr3mlin13, 2024-01-02:
Great Thanks,

I have the basics sorted now, so will not be without power when loadshedding hits. 

Now the fun can start and tinkering to figure everything out properly.

Thanks for the assist, will check back in if I figure someting else out. Hopefully I can assist back sometime in the future.