gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #53: Change guide from Region1 to Region2

CraigBoswell, 2024-02-01:
Apologies if there is a guide already which I have not found, but are there instructions for changes required to move from R1 to R2?
The instructions I find are for new installations, and no sure what should be deleted from existing setup to prepare for R2 setup.
Was really hoping to keep entities with same names etc. so automations can start to flow again.
Any assistance greatly appreciated.
I have a single Sunsynk 8.8kW inverter which was communicating 100% before the Region change.

CraigBoswell, 2024-02-01:
Hi @t0mtaylor thanks for response. 
I decided to delete the old sensor and config files, so the new import did have Region: 2
Only difference to yours is force_short_delay: "false"
Not sure if that makes any difference?

t0mtaylor, 2024-02-01:
@CraigBoswell thats good so you should now be on region 2 👍 

I changed `force_short_delay` key to `true` value to test some debug errors, which occur when you have a plant list containing more than one plant.

if your have a single plant which updates within 60 seconds you can enable this (true), otherwise disable it (false)

```
force_short_delay - If you do not have access to list gateways and you are certain your update time is set to 60 seconds, set to "true"
```

CraigBoswell, 2024-02-01:
Thanks @t0mtaylor , I have had some success this evening and now see data flowing again. Need to do some tidying up, then want to give the automations a go to control the inverter based on loadshedding and/or battery etc., but that will be a next week problem.
Tanks again,

closed 2024-02-01 by CraigBoswell