gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #56: PV3

TanPau1412, 2024-02-11:
Hi,

I am battling to get a 3rd PV string to pulled through. Am I missing something?

**From Dev Tools - States**
sensor.sunsynk_pv3Sunsynk PV3 | unavailable | friendly_name: Sunsynk PV3

**From Sunsynk.net**
Power
1,134.0W
Today
3.6kWh
Total
5,233.2kWh
I-pv-L1
1.8A
V-pv-L1
265.2V
p-pv-L1
490.0W
I-pv-L2
1.6A
V-pv-L2
203.2V
p-pv-L2
335.0W
I-pv-L3
1.5A
V-pv-L3
210.0V
p-pv-L3
309.0W
Grid Tie Power
--W

gdwaterworth, 2024-02-16:
I dont have a PV with 3 strings. If you want to message me your login/pwd to gdwaterworth@gmail.com I can try load it in my test env

closed 2024-02-16 by gdwaterworth

TanPau1412, 2024-02-21:
Hi Gary,Sorry for the late reply but I was having issues with my mail account.The sunsynk username is ***@***.*** and the password is ***@***.*** will see there are multiple plants. The one with the 3 PV strings is WJG 3259Thanks for the great great work you are doingRegards,Tanya