gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #63: Error:  Unable to verify leaf signature

Supermanenvysme, 2024-03-25:
Been experiencing an issue intermittently since this morning. When opening the Sunsynk flow, I get the Error: Unable to verify leaf signature. This appears to be occurring on the http request node.

gdwaterworth, 2024-03-25:
This is on teh Sunsynk API website. Nothing can be fixed in the flows

JGeorgeu, 2024-04-06:
@gdwaterworth just a note on this, I was experiencing this issue as well, and was able to resolve it by updating the header on the request to match the request header when I logged in to the sunsynk portal. In particular, the below worked for me:

`var headers_with_bearer = {
        "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36",
        "Accept": "application/json",
        "Accept-Language": "en-GB,en-US;q=0.9,en;q=0.8",
        "Accept-Encoding": "gzip, deflate, br, zstd",
        "Authorization": "Bearer " + global.get("SS.Bearer"),
        "Origin": "https://sunsynk.net",
        "Referer": "https://sunsynk.net/",
        "Sec-Fetch-Dest": "empty",
        "Sec-Fetch-Mode": "cors",
        "Sec-Fetch-Site": "same-site",
        "Sec-Ch-Ua-Platform": "Windows"
    }`

gdwaterworth, 2024-04-07:
Note on this , headers are still correct , just updated as tested in firefox 

 I have however disabled certificate checking 
Does not help changing to http as sunsynk redirects to https