gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #65: Multiple Inverter Input Issues

NoMhawk, 2024-07-29:
I have just installed and setup HA with your red-node input.

I have 3 inverters (1 master, 2 slaves) the config is set to pull the data from the sunsynk api website.

The issue i'm having is the PV input from the master that is being stored in the "SS_Export_Data.pv.1_power" field is the total combined PV input from all the inverters.

The same happens on the 1_current 

But the voltage its only bringing in the masters PV voltage only as expected.

Is there something I have done wrong?

I have 6 strings (2 on each inverter) I was hoping to at least be able to see the total from each inverter and display that.
Is there a way to see the values from the other inverters?

I can see it finds all 3 inverters and gateways but can't see anywhere where it stores the other data.

Just as a reference the master controls nearly everything settings wise and passes this onto the slaves. Including the battery.

All 3 only show their own individual input and output on the touch screens, but you can see the combined on the api.sunsynk website.

ronnie1102, 2025-02-21:
> I have just installed and setup HA with your red-node input.
>
> I have 3 inverters (1 master, 2 slaves) the config is set to pull the data from the sunsynk api website.
>
> The issue i'm having is the PV [...]

Did you ever get this sorted? Have the same issue at the moment.

ronnie1102, 2025-02-21:
> No I ended up going round and round in circles so got rid of it and used a more simple solution.
> 
> https://github.com/martinville/solarsynkv2

I just had to shutdown my hass instance on my windows laptop and it returned to normal.

Was close to redoing everything, glad you managed.

gdwaterworth, 2025-02-21:
To be honest. I have not looked much at this in a long time. Busy selling this house so may just end up locking the project

https://github.com/martinville/solarsynkv2

https://github.com/martinville/solarsynkv2

ronnie1102, 2025-02-21:
> To be honest. I have not looked much at this in a long time. Busy selling this house so may just end up locking the project
> 
> Sent from Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

Locked or unlocked it’s a really great project as is Gary.

When I first started using it, it was already great.

Was just missing the options to change the time of use and when you launched that, to me it came full circle and could not be happier with your project.

As many others I am sure, I value and appreciate the time and effort taken on this project.