gdwaterworth/Home-Assistant---E-Linter-Logger---Sunsynk #69: Unable to parse API data

Hendrik-LI, 2025-05-17:
Hi @gdwaterworth 

After last week's Sunsynk server outage, my Node Red flow is struggling to process the data from the Sunsynk API.

The flow is running, the Sunsynk credentials are working, the Long Lived HA token is still valid.

Unless done accidentally, no changes have been made to Node Red.

I have not yet tried a fresh re-install (hoping to avoid this).

[image]

[image]

djsly25, 2025-05-17:
Same issue with mine

wacalitz, 2025-05-18:
https://powerforum.co.za/topic/16806-sunsynk-logger-integration-into-home-assistant/page/36/ 
Have look here perhaps this will solve the problem.
This was post HA 2025.5 update this snug in:

"Set Global Vars" 
must be edited to work:

Change the variable headers_with_bearer to ( I changed encoding - removed gzip , referrer and origin )

   var headers_with_bearer = {

        "User-Agent": "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:124.0) Gecko/20100101 Firefox/124.0",

        "Accept": "application/json",

        "Accept-Language": "en-US, en; q=0.5",

        "Accept-Encoding": "deflate",

        "Content-Type": "application/json;charset=UTF-8",

        "Authorization": "Bearer " + global.get("SS.Bearer"),

        "Origin": "https://api.sunsynk.net/",

        "Connection": "keep-alive",

        "Referer": "https://api.sunsynk.net/",

        "Sec-Fetch-Dest": "empty",

        "Sec-Fetch-Mode": "cors",

        "Sec-Fetch-Site": "same-site",

        "TE": "trailers"

    }

Hope this helps.

Hendrik-LI, 2025-05-18:
@wacalitz , thank you both so very much. This solution worked!

@gdwaterworth , thank you for creating this solution.
I was sad to see that you will be stepping away from the project.
Thank you for everything. many many people are in your debt.

closed 2025-05-18 by Hendrik-LI